domoszlai/juicy-gcode #1: Generates INaN JNaN IInfinity JInfinity

ulidtko, 2018-08-07:
I get this "GCode":

```
G00 X-0.2646 Y10.3188
G01 Z0 F10.00
G02 X0.2646 Y10.8479 IInfinity J-Infinity
G02 X0.2646 Y10.8479 INaN JNaN
G00 Z10
G00 X0.2646 Y10.3188
G01 Z0 F10.00
G02 X-0.2646 Y10.8479 IInfinity JInfinity
G02 X-0.2646 Y10.8479 INaN JNaN
G00 Z10
G00 X0.2646 Y10.5833
```

I don't think this is correct in any way.

domoszlai, 2018-08-07:
Agree. Can I have the SVG?

ulidtko, 2018-08-07:
Yes sure. Not minimized though.

[hatching-test.zip](https://github.com/domoszlai/juicy-gcode/files/2265605/hatching-test.zip)

domoszlai, 2018-08-07:
This one causes it for whatever reason. I'll have alook

  <path
     d="M -0.99999997,39.000001 C 0.99999997,41 0.99999997,41 0.99999997,41 m 0,-1.999999 C -0.99999997,41 -0.99999997,41 -0.99999997,41 M 1,40 a 1,0.99999994 0 0 1 -1,1 1,0.99999994 0 0 1 -1,-1 1,0.99999994 0 0 1 1,-1 1,0.99999994 0 0 1 1,1 z"/>

ulidtko, 2018-08-07:
Damn it, Inkscape...

Would it be feasible to uhh *optimize* output arcs with "big enough" radii to line segments then?

The tool trades an approximation withing an error margin, anyway.

domoszlai, 2018-08-07:
I'll add a special case for this

domoszlai, 2018-08-07:
fixed in master, give it a try

domoszlai, 2018-08-08:
v0.1.0.5 has been released

closed 2018-08-08 by domoszlai