domoszlai/juicy-gcode #4: Output mirrored in Y axis?

kcajf, 2020-04-10:
Really enjoying using your tool - runs fast and reliably. Thanks for writing it. The only issue I'm facing is that the output seems to be mirrored relative to the SVG, in the y axis. Is that deliberate? Any way to disable it?

domoszlai, 2020-04-10:
Could you give me an example, please?

kcajf, 2020-04-10:
Sure:

svg:
```
<?xml version="1.0" encoding="utf-8" ?>
<svg xmlns="http://www.w3.org/2000/svg" xmlns:ev="http://www.w3.org/2001/xml-events" xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape" xmlns:xlink="http://www.w3.org/1999/xlink" baseProfile="tiny" height="1000" version="1.2" width="1000">
  <defs/>
  <g fill="none" id="layer1" inkscape:groupmode="layer" inkscape:label="1" stroke="black" style="display:inline">
    <path d="M300.0,300 L600.0,600.0"/>
  </g>
</svg>
```

output gcode:
```
G17
G90
G0 Z10
G0 X0 Y0
M3
G4 P2000.000000
G00 Z10
G00 X105.8333 Y105.8333
G01 Z0 F10.00
G01 X211.6667 Y211.6667
G0 Z10
M5
M2
```

I think the reason is because SVG coordinate system starts at the top left: https://developer.mozilla.org/en-US/docs/Web/SVG/Tutorial/Positions
If you open the svg in an image viewer it should show a line sloping down to the right (though the coordinates in the file are up and to the right.) The gcode line is up and to the right (and the G-Code coordinate system has the origin bottom left).

domoszlai, 2020-04-10:
You're right. The question is how to solve this problem. I just can't invert the Y axis, because other people use it like this. What about a command line option?

kcajf, 2020-04-11:
A command line option would be super!

domoszlai, 2020-05-04:
I made a branch, https://github.com/domoszlai/juicy-gcode/tree/mirror-y, could you test? I can send you a binary if that helps.

ibrahimuslu, 2020-05-11:
Hello,
I have just trying to compile on my Mac but stucked almost 2 hours on the following line. before that everything was fine
```
bifunctors          > Installing library in /Users/mac/.stack/snapshots/x86_64-osx/35b2e985dfa4ae1b89bbc37d1fbd21e6a38abe6ec653894eb825b47bff93b0e8/8.2.2/lib/x86_64-osx-ghc-8.2.2/bifunctors-5.5-BkM1x9B6hjLL8gtezdY6S1
bifunctors          > Registering library for bifunctors-5.5..
Progress 42/51: profunctors, semigroupoids
```

Here is my mac information: 
[image]

Its my first time to compile haskell program. Could you please tell me that if it is normal or not?

domoszlai, 2020-05-11:
I just tried it on a MacBook Pro, same os. It took ~20 minutes to build from scratch, so it is definitely not normal. I use travic-ci to automate Mac builds: https://travis-ci.org/github/domoszlai/juicy-gcode

domoszlai, 2020-05-11:
Attached a Mac binary, does it help? [juicy-gcode.gz](https://github.com/domoszlai/juicy-gcode/files/4609554/juicy-gcode.gz)

ibrahimuslu, 2020-05-11:
It works like a charm 😃 
Here is the results:
The SVG file preview i wanted to convert.  v
[image]

"master" Branch Result: 
[image]

"mirror-y" Branch Result:
[image]

About the compiling and building . I stopped the stucked process and restart. Then it is build successfully in around 20 minutes as you told. 
Thank you very much.

domoszlai, 2020-05-11:
Thank you for testing. It is merged now, I'll try to release it ASAP. What is the software you used to display the GCode BTW?

ibrahimuslu, 2020-05-11:
laserweb

şunu yazdı:

closed 2020-05-11 by domoszlai

kcajf, 2020-05-12:
Thanks @domoszlai !