domoszlai/juicy-gcode #7: Support G5 command

domoszlai, 2020-05-25:
G5 command is going to be supported by GRBL soon, would be nice to be able to generate those from juicy-gcode

domoszlai, 2020-05-25:
A command line argument is added to the `g5` branch: `-b`. When it presents, the beziers are generated as G5 commands instead of being interpolated by arcs.

ithinkido, 2020-05-25:
I have just tried the new G5-branch on the attached file, The output only generates one small section of G05 command.
This is the command that I used : juicy-gcode 2.svg -b -o 2.nc
[G5 test.zip](https://github.com/domoszlai/juicy-gcode/files/4677112/G5.test.zip)
 In the master branch, I was able to use the same file and generated a good match with bi-arcs, so I think the file is ok. 
Am I missing something here ?

domoszlai, 2020-05-25:
Oh, I forgot the recursive step... Could you try it again please?

ithinkido, 2020-05-25:
That looks perfect ! Can we leave this open for a while, I still need to test this some more ( on real hardware )

domoszlai, 2020-05-25:
If it works in simulation, it should be OK on real hardware as well. I'm
going to merge the `g5` branch to master and make a new release.

domoszlai, 2020-05-25:
A small warning: you get upside down gcode unless you use `-m`

ithinkido, 2020-05-25:
Ok , thanks for the warning.

ithinkido, 2020-05-26:
a quick simulation
[image]

domoszlai, 2020-05-26:
Nice! I close this issue now. The `g5` branch will be merged to master and released, so the feature will be available later

closed 2020-05-26 by domoszlai