domoszlai/juicy-gcode #8: Floating point dpi values

ibrahimuslu, 2020-06-23:
Hello, 
some svg files has dpi value 25.4 because of exactly created by inch measurements

but dpi parameter doesnt except floating values. 

I have tried to fix it but got error at last after several trial steps:
`In a pattern binding: (w, h) = SVG.documentSize dpi doc` 

can you tell me if it can be done or not ? 
Thanks in advance.

domoszlai, 2020-06-23:
I've never heard about non integer dpis. What is the software that generates this? Could you please share one of these SVGs?

ibrahimuslu, 2020-06-23:
Here is the sample svg file which dpi supposed to be floating number 25.4
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!-- Created with Inkscape (http://www.inkscape.org/) -->

<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   width="210mm"
   height="297mm"
   viewBox="0 0 210 297"
   version="1.1"
   id="svg8"
   inkscape:version="0.92.3 (2405546, 2018-03-11)"
   sodipodi:docname="square100.svg">
  <defs
     id="defs2" />
  <sodipodi:namedview
     id="base"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:pageopacity="0.0"
     inkscape:pageshadow="2"
     inkscape:zoom="1"
     inkscape:cx="18.107968"
     inkscape:cy="422.7382"
     inkscape:document-units="mm"
     inkscape:current-layer="layer1"
     showgrid="false"
     inkscape:window-width="3200"
     inkscape:window-height="1721"
     inkscape:window-x="-9"
     inkscape:window-y="-9"
     inkscape:window-maximized="1" />
  <metadata
     id="metadata5">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <g
     inkscape:label="Laag 1"
     inkscape:groupmode="layer"
     id="layer1">
    <path
       style="opacity:1;fill:none;fill-opacity:1;stroke:#000000;stroke-width:0.1;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1"
       d="M 0.0,100.0 h 100.0 v -100.0 h -100.0 Z"
       id="rect4518"
       inkscape:connector-curvature="0" />
  </g>
</svg>
```

domoszlai, 2020-06-23:
The size of the canvas is explicitly set in mm and the viewbox is also present. The Dpi setting only matters if the Svg is completely unit less. In this case, setting the dpi wouldn't make any difference in the gcode.

domoszlai, 2020-06-23:
At least it should not matter, if it makes a difference, that's a bug. Maybe the documentation of that command line argument could be better...

domoszlai, 2020-06-23:
If you change
```
width="210mm"
height="297mm"
viewBox="0 0 210 297"
```
to
```
width="210"
height="297"
```
then the Dpi should be 25.4 to have the same size, but I doubt that you can generate something like that, or that's valid at all. 

I saw before unit less Inkscape svg's before, but the dpi was always 96, I'm not sure if that can be changed.

ibrahimuslu, 2020-06-24:
So may be unit conversion option argument could be better ? What do you think?

domoszlai, 2020-06-24:
Yes. that might be better. I can't break compatibility now, so I'm going to change the wording of the help message, but I'm planning to overhaul some stuff and release a new major version. I'll change this argument then.

domoszlai, 2020-06-26:
Is this better?

```
Available options:
  -h,--help                Show this help text
  SVGFILE                  The SVG file to be converted
  -f,--flavor CONFIGFILE   Configuration of G-Code flavor
  -o,--output OUTPUTFILE   The output G-Code file (default is standard output)
  -d,--dpi DPI             Dot per inch; used to determine the size of the SVG
                           when it does not contain any units (default is 72)
  -m,--mirror-y-axis       Mirror Y axis to have the result in G-Code coordinate
                           system
  -b,--generate-bezier     Generate bezier curves (G5) instead of arcs (G2,G3)
```

Probably it should also be called PPI instead of DPI, it is a bit confusing in the context of vector graphics. But I can't change that now anyway.

domoszlai, 2020-06-26:
Inkscape also calls it DPI, so maybe I'll rename like that: `default-resolution` (in DPI)

https://wiki.inkscape.org/wiki/index.php/Release_notes/0.92#Important_changes

ibrahimuslu, 2020-07-08:
thats true. For this problem i found a workaround solution. I am giving a scale of 10 and give the dpi value 254 so i have achieved the exact measures values. 
I dont think there is a confusing dpi or ppi or in description but some of the svgs are drawn in the real measurements so the dpi value should be the same as inch conversion
If you have a chance to put an argument to select ppm or ppi so that instead of inch conversion it will be the mm conversion so there is no need to give workaround solution but only an integer value. as an example instead 25.4, 10 is covers the solution.

domoszlai, 2020-07-08:
I still don't understand why do you want to specify the DPI. The SVG you attached contains unit, so you don't need to

ibrahimuslu, 2020-07-08:
without giving any dpi value while converting, the result is:
```
./juicy-gcode /Users/mac/Documents/CADCAM/100test.svg 
G17
G90
G0 Z10
G0 X0 Y0
M3
G4 P2000.000000
G00 Z10
G00 X0.0000 Y35.2778
G01 Z0 F10.00
G01 X35.2778 Y35.2778
G01 X35.2778 Y0.0000
G01 X0.0000 Y0.0000
G01 X0.0000 Y35.2778
G0 Z10
M5
M2
```

But what i want is:
```
G17
G90
G0 Z10
G0 X0 Y0
M3
G4 P2000.000000
G00 Z10
G00 X0.0000 Y100
G01 Z0 F10.00
G01 X100 Y100
G01 X100 Y0.0000
G01 X0.0000 Y0.0000
G01 X0.0000 Y100
G0 Z10
M5
M2
```

for the second result the dpi value should be given to the juicy-code: 25.4

domoszlai, 2020-07-08:
This is the same SVG that you shared before? If not, could you please share it?

ibrahimuslu, 2020-07-08:
> This is the same SVG that you shared before? If not, could you please share it?

Definitely the same SVG file i have shared before. The following one.

> Here is the sample svg file which dpi supposed to be floating number 25.4
> 
> ```
> <?xml version="1.0" encoding="UTF-8" standalone="no"?>
> 
> 
> <svg
>    xmlns:dc="http://purl.org/dc/elements/1.1/"
>    xmlns:cc="http://creativecommons.org/ns#"
>    xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
>    xmlns:svg="http://www.w3.org/2000/svg"
>    xmlns="http://www.w3.org/2000/svg"
>    xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
>    xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
>    width="210mm"
>    height="297mm"
>    viewBox="0 0 210 297"
>    version="1.1"
>    id="svg8"
>    inkscape:version="0.92.3 (2405546, 2018-03-11)"
>    sodipodi:docname="square100.svg">
>   <defs
>      id="defs2" />
>   <sodipodi:namedview
>      id="base"
>      pagecolor="#ffffff"
>      bordercolor="#666666"
>      borderopacity="1.0"
>      inkscape:pageopacity="0.0"
>      inkscape:pageshadow="2"
>      inkscape:zoom="1"
>      inkscape:cx="18.107968"
>      inkscape:cy="422.7382"
>      inkscape:document-units="mm"
>      inkscape:current-layer="layer1"
>      showgrid="false"
>      inkscape:window-width="3200"
>      inkscape:window-height="1721"
>      inkscape:window-x="-9"
>      inkscape:window-y="-9"
>      inkscape:window-maximized="1" />
>   <metadata
>      id="metadata5">
>     <rdf:RDF>
>       <cc:Work
>          rdf:about="">
>         <dc:format>image/svg+xml</dc:format>
>         <dc:type
>            rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
>         <dc:title></dc:title>
>       </cc:Work>
>     </rdf:RDF>
>   </metadata>
>   <g
>      inkscape:label="Laag 1"
>      inkscape:groupmode="layer"
>      id="layer1">
>     <path
>        style="opacity:1;fill:none;fill-opacity:1;stroke:#000000;stroke-width:0.1;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1"
>        d="M 0.0,100.0 h 100.0 v -100.0 h -100.0 Z"
>        id="rect4518"
>        inkscape:connector-curvature="0" />
>   </g>
> </svg>
> ```

domoszlai, 2020-07-08:
I believe it's a bug that the Dpi settings affects the result. I'll have a
look

domoszlai, 2020-07-08:
Are you using the latest version?

I just tried and it produces the same gcode whatever dpi I provide (exactly as it should work):

```
c:\Users\dlacko\Documents\projects\juicy-gcode>stack run -- test.svg
G17
G90
G0 Z10
G0 X0 Y0
M3
G4 P2000.000000
G00 Z10
G00 X0.0000 Y100.0000
G01 Z0 F10.00
G01 X100.0000 Y100.0000
G01 X100.0000 Y0.0000
G01 X0.0000 Y0.0000
G01 X0.0000 Y100.0000
G0 Z10
M5
M2

c:\Users\dlacko\Documents\projects\juicy-gcode>stack run -- test.svg -d 10
G17
G90
G0 Z10
G0 X0 Y0
M3
G4 P2000.000000
G00 Z10
G00 X0.0000 Y100.0000
G01 Z0 F10.00
G01 X100.0000 Y100.0000
G01 X100.0000 Y0.0000
G01 X0.0000 Y0.0000
G01 X0.0000 Y100.0000
G0 Z10
M5
M2

c:\Users\dlacko\Documents\projects\juicy-gcode>stack run -- test.svg -d 30
G17
G90
G0 Z10
G0 X0 Y0
M3
G4 P2000.000000
G00 Z10
G00 X0.0000 Y100.0000
G01 Z0 F10.00
G01 X100.0000 Y100.0000
G01 X100.0000 Y0.0000
G01 X0.0000 Y0.0000
G01 X0.0000 Y100.0000
G0 Z10
M5
M2
```

domoszlai, 2020-08-07:
Could you please confirm that it works with the latest version? (working = dpi setting has no effect on the generated gcode)

domoszlai, 2020-08-19:
I close this ticket, if you still experience the problem, please open a new one.

closed 2020-08-19 by domoszlai