domoszlai/juicy-gcode #9: Some NaN values generated

ibrahimuslu, 2020-07-08:
Hello, 
for some of the svg files, generated NaN values for the G2 or G3 

Can we overcome with this solution via constraint a conversion limit or sth. like that. 
```

G01 X8.3642 Y53.2708
G01 X8.3748 Y53.2785
G02 XNaN YNaN INaN JNaN
G02 X8.3748 Y53.2979 INaN JNaN
```

[Archive.zip](https://github.com/domoszlai/juicy-gcode/files/4889489/Archive.zip)*
*There is testtext.svg and testtext.gcode files in this archive

I cant really exactly understand how this code generating g2 conversion. There is numerical solution with iterations as i understand but it is really hard for to understand how it is implemented in haskell. I have tried to solve but i cant manage to do. 
If you can help, it will be really really big favor. 
Thanks in advance

domoszlai, 2020-07-08:
I can reproduce the problem, I will have a look

domoszlai, 2020-08-08:
Branch [approx-freeze](https://github.com/domoszlai/juicy-gcode/tree/approx-freeze) fixes your issue. It also 
introduces a parameter, called RESOLUTION, to control the approximation. Would you be able to test it from the branch?

domoszlai, 2020-08-19:
Juicy-GCode v0.1.0.10 has been released and that version fixes your this issue. I close this ticket, if you still experience the problem, please open a new one.

closed 2020-08-19 by domoszlai