domoszlai/juicy-gcode #10: Tag <line> is not parsed properly

berkut0, 2020-08-05:
```xml
<?xml version="1.0" encoding="utf-8"?>
<!-- Generator: Adobe Illustrator 24.1.1, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->
<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 width="850.4px" height="850.4px" viewBox="0 0 850.4 850.4" enable-background="new 0 0 850.4 850.4" xml:space="preserve">
<rect x="141.7" y="141.7" fill="none" stroke="#231F20" stroke-miterlimit="10" width="566.9" height="566.9"/>
<circle fill="none" stroke="#231F20" stroke-miterlimit="10" cx="425.2" cy="425.2" r="141.7"/>
<line fill="none" stroke="#231F20" stroke-miterlimit="10" x1="525.4" y1="525.4" x2="708.7" y2="708.7"/>
<line fill="none" stroke="#231F20" stroke-miterlimit="10" x1="141.7" y1="141.7" x2="325" y2="325"/>
<line fill="none" stroke="#231F20" stroke-miterlimit="10" x1="325" y1="525.4" x2="141.7" y2="708.7"/>
<line fill="none" stroke="#231F20" stroke-miterlimit="10" x1="708.7" y1="141.7" x2="525.4" y2="325"/>
<line fill="none" stroke="#231F20" stroke-miterlimit="10" x1="141.7" y1="425.2" x2="708.7" y2="425.2"/>
<line fill="none" stroke="#231F20" stroke-miterlimit="10" x1="425.2" y1="141.7" x2="425.2" y2="708.7"/>
</svg>
```

this parsed in:

<details>

```gcode
G00 Z9 F800
G00 X49.9886 Y49.9886
G01 Z7.5 F800
G01 X249.9783 Y49.9886
G01 X249.9783 Y249.9783
G01 X49.9886 Y249.9783
G01 X49.9886 Y49.9886
G00 Z9 F800
G00 X100.0125 Y150.0011
G01 Z7.5 F800
G03 X100.6705 Y141.8762 I50.4891 J0.0000
G03 X102.6151 Y134.0391 I48.8960 J7.9725
G03 X113.7950 Y115.5156 I47.3541 J15.9451
G03 X131.7707 Y103.4409 I36.2368 J34.5284
G03 X140.6842 Y100.8810 I18.0648 J46.1019
G03 X150.0011 Y100.0125 I9.3170 J49.5426
G03 X158.1261 Y100.6705 I0.0000 J50.4891
G03 X165.9631 Y102.6151 I-7.9725 J48.8960
G03 X184.4866 Y113.7950 I-15.9451 J47.3541
G03 X196.5613 Y131.7707 I-34.5284 J36.2368
G03 X199.1213 Y140.6842 I-46.1019 J18.0648
G03 X199.9897 Y150.0011 I-49.5426 J9.3170
G03 X199.3317 Y158.1261 I-50.4891 J0.0000
G03 X197.3871 Y165.9631 I-48.8960 J-7.9725
G03 X186.2072 Y184.4866 I-47.3541 J-15.9451
G03 X168.2315 Y196.5613 I-36.2368 J-34.5284
G03 X159.3181 Y199.1213 I-18.0648 J-46.1019
G03 X150.0011 Y199.9897 I-9.3170 J-49.5426
G03 X141.8762 Y199.3317 I0.0000 J-50.4891
G03 X134.0391 Y197.3871 I7.9725 J-48.8960
G03 X115.5156 Y186.2072 I15.9451 J-47.3541
G03 X103.4409 Y168.2315 I34.5284 J-36.2368
G03 X100.8810 Y159.3181 I46.1019 J-18.0648
G03 X100.0125 Y150.0011 I49.5426 J-9.3170
G00 Z9 F800
G00 X185.3494 Y185.3494
G01 Z7.5 F800
G01 X185.3494 Y185.3494
G00 Z9 F800
G00 X49.9886 Y49.9886
G01 Z7.5 F800
G01 X49.9886 Y49.9886
G00 Z9 F800
G00 X114.6528 Y185.3494
G01 Z7.5 F800
G01 X114.6528 Y185.3494
G00 Z9 F800
G00 X250.0136 Y49.9886
G01 Z7.5 F800
G01 X250.0136 Y49.9886
G00 Z9 F800
G00 X49.9886 Y150.0011
G01 Z7.5 F800
G01 X49.9886 Y150.0011
G00 Z9 F800
G00 X150.0011 Y49.9886
G01 Z7.5 F800
G01 X150.0011 Y49.9886
G0 Z10
```
</details>

domoszlai, 2020-08-07:
It is indeed wrong. I'll have a look

domoszlai, 2020-08-07:
Fixed on master. I'll have a look at your other issue, then I release them together. Thank you for reporting it

closed 2020-08-07 by domoszlai