domoszlai/juicy-gcode #11: Some files cause a freeze

berkut0, 2020-08-05:
<details><summary><b>This svg code cause a freeze</b></summary>

```xml

<?xml version="1.0" encoding="utf-8"?>
<!-- Generator: Adobe Illustrator 24.1.1, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->
<svg version="1.1" id="Layer_2_1_" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 width="850.4px" height="948.4px" viewBox="0 0 850.4 948.4" enable-background="new 0 0 850.4 948.4" xml:space="preserve">
<g>
	<g>
		<g>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M593.1,518.8l-0.5,0.3L592,519
				c0,0,0.2-0.6,0.3-1.2c0.2-0.6,0.3-1.2,0.3-1.2l-0.2-0.3l0.1-0.4l1.4,0.3c0,0,0.2-0.8,0.5-2s0.6-2.7,0.9-4.3
				c0.3-1.6,0.6-3.1,0.8-4.3c0.1-0.6,0.2-1.1,0.2-1.4c0.1-0.3,0.1-0.5,0.1-0.5l-0.3-0.4l0.1-0.7l0.6,0.9c0,0,0,0.2-0.1,0.6
				s-0.1,0.9-0.3,1.5c-0.1,0.6-0.3,1.4-0.4,2.2c-0.1,0.8-0.3,1.7-0.5,2.5c-0.4,1.7-0.7,3.4-1,4.7s-0.5,2.1-0.5,2.1s-0.1,0.5-0.2,1
				c-0.1,0.5-0.2,1-0.2,1L593.1,518.8z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M590.3,534.6c0,0,0.2-0.4,0.4-1.1
				s0.6-1.8,1-3s0.8-2.7,1.3-4.3c0.4-1.6,0.9-3.2,1.4-4.9c0.4-1.7,0.9-3.4,1.2-5c0.2-0.8,0.4-1.6,0.5-2.3c0.2-0.7,0.3-1.4,0.4-2.1
				c0.3-1.3,0.5-2.3,0.6-3.1c0.1-0.7,0.2-1.2,0.2-1.2l0.3,0.5c0,0-0.1,0.4-0.2,1.2c-0.1,0.7-0.4,1.8-0.6,3.1
				c-0.5,2.6-1.3,5.9-2.2,9.3c-0.5,1.7-0.9,3.4-1.4,4.9c-0.5,1.6-0.9,3-1.3,4.3c-0.4,1.2-0.8,2.3-1,3c-0.1,0.4-0.2,0.6-0.3,0.8
				s-0.1,0.3-0.1,0.3L590.3,534.6z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M603.3,518.3c0,0-0.4-0.7-0.9-1.7
				c-0.3-0.5-0.6-1.1-1-1.7c-0.4-0.6-0.8-1.3-1.2-1.9c-0.8-1.3-1.6-2.6-2.2-3.6c-0.6-1-1-1.6-1-1.6s0.1-0.8,0.4-2
				c0.2-1.2,0.4-2.8,0.7-4.4c0.2-1.6,0.4-3.2,0.6-4.4c0.1-1.2,0.2-2,0.2-2l1.2-1c0,0,0.1-0.4,0.1-1.1c0.1-0.7,0.1-1.6,0.2-2.4
				c0.1-0.9,0.1-1.8,0.2-2.4c0-0.7,0.1-1.1,0.1-1.1l5,0.3c0,0,0,0.8-0.1,2s-0.2,2.8-0.4,4.4s-0.3,3.2-0.5,4.4s-0.3,2-0.3,2
				s0.8,1.1,1.6,2.3c0.8,1.1,1.5,2.3,1.5,2.3s-0.1,0.7-0.3,1.8s-0.4,2.5-0.7,3.9c-0.2,1.4-0.6,2.8-0.8,3.9s-0.4,1.8-0.4,1.8
				l-0.8,0.6l-0.5,2.1L603.3,518.3z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M602.5,515c0,0,0.3-1.3,0.7-3.4
				c0.2-1,0.4-2.2,0.6-3.5c0.2-1.3,0.5-2.6,0.7-4c0.9-5.4,1.4-10.9,1.4-10.9s-1-1.3-2-2.5c-0.5-0.6-1-1.2-1.4-1.7s-0.6-0.8-0.6-0.8
				l-9.1-0.6c0,0,0,0.3,0,0.7s-0.1,1-0.1,1.5c-0.1,1.1-0.2,2.2-0.2,2.2l-2.6-0.2c0,0-0.2,0.2-0.5,0.4s-0.6,0.5-1,0.8
				c-0.7,0.6-1.5,1.2-1.5,1.2l-10.2-1.1l-1.3,1l-3.6-0.4c0,0-1-1.1-1.9-2.1c-0.5-0.5-1-1-1.3-1.4c-0.4-0.4-0.6-0.6-0.6-0.6l-2-0.2
				l-1.3-1.3l0.1-0.8l7.9,0.6l1.7,1.9c0,0,0,0.6-0.1,1.2s-0.1,1.2-0.1,1.2l7.9,0.8l0.2-1.4c0,0,0.6-0.5,1.2-1c0.6-0.5,1.2-1,1.2-1
				l6.8,0.6c0,0,0-0.5,0.1-1.2s0.1-1.7,0.2-2.6c0.1-0.9,0.1-1.9,0.1-2.6s0-1.2,0-1.2l1.8-1.7l2.8,0.1l-0.1,2.3l2,0.1
				c0,0,1.2-1.2,2.4-2.3c1.2-1.2,2.3-2.4,2.3-2.4l12.8,0.1c0,0,0,0.8,0,1.6s0,1.6,0,1.6l-2.2,2.3l-3.9-0.2c0,0,0,0.2,0,0.5
				s0,0.8-0.1,1.3c-0.1,1.1-0.2,2.6-0.2,4c-0.1,1.5-0.2,2.9-0.3,4c0,0.5-0.1,1-0.1,1.3s-0.1,0.5-0.1,0.5s-0.7,0.6-1.4,1.2
				s-1.4,1.2-1.4,1.2s0,0.3-0.1,0.7c-0.1,0.5-0.2,1.2-0.3,2c-0.2,1.6-0.6,3.8-0.9,6c-0.2,1.1-0.4,2.2-0.6,3.2
				c-0.1,0.5-0.2,1-0.3,1.5c-0.1,0.5-0.2,0.9-0.3,1.3c-0.3,1.6-0.6,2.7-0.6,2.7L602.5,515z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M592.6,490.9l8,0.7c0,0,0.5,0.6,1,1.2
				c0.2,0.3,0.5,0.6,0.7,0.9c0.2,0.2,0.3,0.4,0.3,0.4l2.3,0.2c0,0-0.1,1-0.2,2c-0.1,0.5-0.1,1-0.2,1.4c0,0.4-0.1,0.6-0.1,0.6l-1-0.1
				c0,0-0.1,0.7-0.2,1.7c-0.1,0.5-0.1,1.1-0.2,1.7c-0.1,0.6-0.2,1.3-0.3,2c-0.2,1.3-0.4,2.7-0.6,3.7c-0.2,1-0.3,1.7-0.3,1.7l0.6,1
				l1.1,0.2c0,0,0.2-1.1,0.5-2.6c0.2-0.8,0.3-1.7,0.4-2.7c0.2-1,0.3-2,0.5-3.1c0.3-2.1,0.6-4.2,0.7-5.8c0.1-0.8,0.2-1.5,0.2-1.9
				c0.1-0.5,0.1-0.7,0.1-0.7s-1-1.3-2-2.5c-0.5-0.6-1-1.2-1.4-1.7s-0.6-0.8-0.6-0.8l-9.1-0.6c0,0,0,0.8-0.1,1.6
				C592.6,490.1,592.6,490.9,592.6,490.9z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M614.8,480.9l-1.2,1.3l-12.1-0.4
				c0,0,0.6-0.6,1.1-1.1c0.6-0.6,1.1-1.2,1.1-1.2l11.1,0.2c0,0,0,0.3,0,0.6C614.8,480.7,614.8,480.9,614.8,480.9z"/>
			<polygon fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" points="607.5,481.7 606,483.2 601.5,483.1 
				601.5,481.9 			"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M609.3,483.7c0,0,0,0.6-0.1,1.5
				s-0.1,2.2-0.2,3.4c-0.1,1.2-0.2,2.5-0.3,3.4s-0.1,1.5-0.1,1.5l-0.8,0.7l-0.8-0.1c0,0,0.1-0.6,0.1-1.4c0.1-0.8,0.2-2,0.3-3.1
				s0.1-2.3,0.2-3.1c0-0.8,0.1-1.4,0.1-1.4l-0.6-0.8v-0.9L609.3,483.7z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M599.8,493.1l-3-0.3c0,0-0.1,0.6-0.2,1.4
				c-0.1,0.9-0.2,2-0.4,3.2c-0.2,1.1-0.3,2.3-0.4,3.2c-0.1,0.4-0.1,0.8-0.2,1c0,0.3-0.1,0.4-0.1,0.4l1.4,2c0,0-0.2,1.3-0.5,3.2
				s-0.8,4.4-1.4,6.9c-0.3,1.3-0.6,2.5-0.9,3.7c-0.3,1.2-0.6,2.3-0.8,3.2s-0.5,1.7-0.6,2.3c-0.2,0.5-0.2,0.9-0.2,0.9L592,524
				c0,0-0.2,0.6-0.3,1.1c-0.2,0.6-0.3,1.1-0.3,1.1l-0.4-0.1l-0.5,1.5l0.4,0.8c0,0-0.1,0.2-0.1,0.5c-0.1,0.3-0.2,0.7-0.4,1.3
				c-0.4,1-0.8,2.4-1.3,3.8c-0.5,1.4-1,2.7-1.4,3.8c-0.4,1-0.7,1.7-0.7,1.7s0.5,1.2,0.9,2.3c0.2,0.6,0.4,1.2,0.6,1.6
				c0.2,0.4,0.3,0.7,0.3,0.7l1.6,0.6c0,0-0.5,1.1-0.9,2.2c-0.2,0.6-0.5,1.1-0.7,1.5s-0.3,0.7-0.3,0.7l0.5,1.5l2.5,1.1l0.4-0.8
				l1.9,0.9l-0.4,0.8l1.3,0.6c0,0,0.2-0.4,0.5-1s0.6-1.4,1-2.2c0.7-1.6,1.4-3.3,1.4-3.3l-2.3-0.9c0,0-0.1-0.2-0.2-0.5
				c-0.1-0.3-0.3-0.8-0.5-1.4c-0.5-1.2-1.1-2.8-1.7-4.3c-0.6-1.6-1.3-3.1-1.8-4.2c-0.5-1.1-0.8-1.9-0.8-1.9s0.1-0.4,0.4-1.1
				c0.2-0.7,0.6-1.8,1-3c0.8-2.5,1.8-5.9,2.6-9.2c0.2-0.8,0.4-1.7,0.6-2.5c0.2-0.8,0.4-1.7,0.6-2.5c0.1-0.4,0.2-0.8,0.3-1.2
				s0.2-0.8,0.2-1.1c0.2-0.7,0.3-1.4,0.4-2.1c0.3-1.3,0.4-2.4,0.6-3.1c0.1-0.8,0.2-1.2,0.2-1.2s0.5-0.4,1-0.7c0.5-0.4,1-0.8,1-0.8
				s0.1-0.7,0.3-1.7c0.1-0.5,0.2-1.1,0.3-1.7s0.2-1.3,0.3-2s0.2-1.3,0.3-2c0.1-0.6,0.1-1.2,0.2-1.7c0.1-1,0.2-1.7,0.2-1.7
				L599.8,493.1z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M592.1,537.7c0,0,0.3-1,0.9-2.4
				c0.3-0.7,0.5-1.6,0.8-2.5s0.6-1.9,0.9-2.8c0.3-1,0.6-1.9,0.9-2.9c0.3-0.9,0.5-1.8,0.7-2.5s0.4-1.3,0.5-1.8
				c0.1-0.4,0.2-0.7,0.2-0.7l-1.4-2.6l-1.3-0.3c0,0,0.2-0.6,0.3-1.2c0.1-0.3,0.1-0.6,0.2-0.8c0.1-0.2,0.1-0.4,0.1-0.4l2.6,0.6
				l1.3-0.9l1.3,0.3l-0.5,2.1l-0.8,0.5c0,0-0.2,0.6-0.4,1.5s-0.5,2.1-0.9,3.3c-0.3,1.2-0.7,2.4-0.9,3.3c-0.3,0.9-0.5,1.5-0.5,1.5
				l-0.6,0.4c0,0-0.1,0.5-0.4,1.2c-0.2,0.7-0.5,1.6-0.9,2.6c-0.3,0.9-0.6,1.9-0.9,2.6c-0.3,0.7-0.4,1.2-0.4,1.2s0.5,1.1,0.9,2.2
				c0.2,0.5,0.4,1.1,0.6,1.5c0.2,0.4,0.3,0.7,0.3,0.7l-0.6,1.5c0,0-0.5-1.3-1-2.6c-0.3-0.6-0.5-1.3-0.7-1.8
				C592.3,538,592.1,537.7,592.1,537.7z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M594.5,516.8c0,0,0,0.2-0.1,0.4
[... 175 lines cut ...]
				c0.9,0.9,1.9,2.1,3,3.2c1.1,1.2,2.2,2.5,3.4,3.8c1.1,1.3,2.2,2.7,3.3,3.9c1,1.3,2,2.5,2.8,3.5c1.6,2.1,2.6,3.5,2.6,3.5l1.2-0.9
				c0,0-1.1-1.4-2.6-3.5c-0.8-1-1.8-2.2-2.8-3.5s-2.2-2.6-3.3-3.9c-1.2-1.3-2.3-2.6-3.4-3.8c-1.1-1.2-2.1-2.3-3-3.3
				c-1.8-1.9-3-3.1-3-3.1L555.8,345.7z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M553.8,347.6c0,0,1.2,1.2,3,3.1
				c0.9,0.9,1.9,2,3,3.2s2.2,2.5,3.3,3.7c1.1,1.3,2.2,2.6,3.2,3.8c1,1.3,1.9,2.4,2.7,3.4c1.5,2,2.6,3.4,2.6,3.4l0.5-0.4
				c0,0-1-1.4-2.6-3.4c-0.8-1-1.7-2.2-2.7-3.4s-2.1-2.6-3.2-3.9c-1.1-1.3-2.3-2.6-3.3-3.8c-1.1-1.2-2.1-2.3-3-3.2
				c-1.8-1.8-3-3.1-3-3.1L553.8,347.6z"/>
		</g>
		<g>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M574.8,368.1c0,0,0.8,1.1,1.6,2.2
				c0.4,0.6,0.8,1.1,1.1,1.5s0.5,0.7,0.5,0.7l1.2-0.8c0,0-0.2-0.3-0.5-0.7c-0.3-0.4-0.7-1-1.1-1.6c-0.8-1.1-1.6-2.2-1.6-2.2
				L574.8,368.1z"/>
			<path fill="none" stroke="#231F20" stroke-width="0.2835" stroke-miterlimit="10" d="M572.6,369.7c0,0,0.8,1.1,1.6,2.2
				c0.4,0.5,0.8,1.1,1,1.5c0.3,0.4,0.5,0.7,0.5,0.7l0.5-0.3c0,0-0.2-0.3-0.5-0.7c-0.3-0.4-0.7-1-1.1-1.5c-0.8-1.1-1.6-2.2-1.6-2.2
				L572.6,369.7z"/>
		</g>
	</g>
</g>
</svg>

```

</details>

It can be rendered with -b command, but because of #10 (or something else) full file isn't drawing correctly.

domoszlai, 2020-08-07:
I saw a similar problem at #9 , my guess is that the algorithm does not converge for whatever reason. I'll have a look

domoszlai, 2020-08-08:
Branch [approx-freeze](https://github.com/domoszlai/juicy-gcode/tree/approx-freeze) fixes your issue. It also 
introduces a parameter, called RESOLUTION, to control the approximation. Would you be able to test it from the branch?

berkut0, 2020-08-10:
Looks like this is not freezing anymore, later I can try to draw the resulting g-code.

berkut0, 2020-08-10:
Maybe it's a little off topic, but maybe you should note this tools written in Rust:
https://github.com/RazrFalcon/resvg/tree/master/usvg

berkut0, 2020-08-19:
I made two tests with huge files with difficult small details and it looks like it works perfect
but now I noticed that there is no a tooloff comand at the end, actually it's a very strange move which needs some investigation

[image]

domoszlai, 2020-08-19:
Could you please explain why it is unexpected?

domoszlai, 2020-08-19:
FYI: Juicy-GCode v0.1.0.10 has been released with the stability fix

berkut0, 2020-08-19:
[image]

[image]

<details>
  <summary>code respectively</summary>

[image]

[image]

</details>

I know, this is a typo at the end, but anyway. Looks like the last movement was not contained in the original file.
Z2 — tooloff, Z-0.5 — toolon

domoszlai, 2020-08-19:
Could you attach the SVG file as well, please?

domoszlai, 2020-08-19:
If Z2 is tooloff, then how come that the G00 after that draws something?

berkut0, 2020-08-19:
Last row is the "end" part of the flavor

But another file just finished printing and it's perfect except that i'm explicitly put neccesary code at the end block

[image]
[image]

domoszlai, 2020-08-19:
I'm a bit confused. The problem is that the software does not generates a tooloff after the last drawing operation?

berkut0, 2020-08-19:
Yes, for now, after all investigations, it seems like left only one problem. I don't know what it was, but I sure that it was my mistake.
If I see anything in the future, I’ll let you know.

domoszlai, 2020-08-19:
Great! Then I close the ticket. Thank you for contributing

closed 2020-08-19 by domoszlai