domoszlai/juicy-gcode #15: Quadratic spline

kimstik, 2022-07-28:
Since cubic G5 is working fine, may we extend support to G5.1 (and even scared to say perhaps G5.2)?

domoszlai, 2022-07-30:
I don't see why not, at least the 5.1. However, internally, beziers are stored as cubic, so we need to generate two quadratic for all of them. What would be the reason for 5.1?

kimstik, 2022-07-30:
Thanks,
the reason is that 5.1 is supported by linuxcnc and grblHAL
if some path can be well interpolated well by a quadratic, why not?
question of freedom :)

domoszlai, 2022-07-31:
You mean that linuxcnc and grblhal supports 5.1, but not 5?

kimstik, 2022-07-31:
actually both of them supports 5 and 5.1
grblHAL does not support 5.2 yet

domoszlai, 2022-07-31:
all right. so 5.1 probably can be done easily, maybe some internal refactoring would be nice, so we actually be able to save on something. but, because these software support vanilla 5, I think there is no urgency

kimstik, 2022-07-31:
sure. no stress ;) just fun