domoszlai/juicy-gcode #16: "NaN" issue with arcs

berkut0, 2022-10-21:
[Uploading vvvv_zebra.zip…]()

[image]

domoszlai, 2022-10-25:
Thanks for reporting it. I'll have a look

domoszlai, 2022-10-26:
May I have the original SVG?

berkut0, 2022-10-28:
Oh, thought I put it in the archive

[zebra_svg_gcodes.zip](https://github.com/domoszlai/juicy-gcode/files/9886841/zebra_svg_gcodes.zip)

domoszlai, 2022-10-29:
This is one of the problematic beziers. Seems impossible
`<path d="M443 1265 C442 1265 440 1266 440 1265" stroke="#ff0000"></path>`

berkut0, 2022-10-31:
A very rare error, difficult to catch, if it happens again, I'll write. I had only a couple of files like this, now I've lost track of which ones.

domoszlai, 2022-10-31:
All right. Your zebra svg works properly now, I'm going to release

closed 2022-10-31 by domoszlai