domoszlai/juicy-gcode #17: Series of Points instead of arcs (Feature Request)

berkut0, 2022-10-29:
There are couple important aspects to this approach:

Many consumer CNCs require recompiling the firmware to use arcs. For example, I had to recompile the marlin firmware of my 3d printer to run the resulting g-codes.

There are often problems when reproducing arcs that are too small. I am now experimenting with robots and using g-codes to convert to commands directly on the robot and too small values fall under calculation problems

domoszlai, 2022-10-29:
I'm wondering what happens if you get unconnected dots. It is just not possible for the device to figure out the proper order to follow them imho.

It also feels like 2 different problems:
1. If your device does not support arcs, we could fall back to straight lines? 
2. What about having a minimal arc/line length setting?

berkut0, 2022-10-29:
G02/G03 clearly indicate the direction of travel
if you look at how the printer firmware is organized, you can specify the level of sampling within the firmware itself
it seems to me it is possible to specify the sampling angle, for example
the G00/G01 series instead of arcs can greatly reduce the risk of getting an error in some "unprepared" setups

domoszlai, 2022-10-29:
Let step back a bit.  One problem is that some devices do not support arcs out of box. That basically means falling back two G01/G02 what is linear approximation. 
> 3D printers most often do not generate arcs, but make a set of points on a circle. 
This is linear approximation imho

The other problem is arcs being too small. In my opinion, it is an approximation problem, probably the error is set too small. What if you change the  --resolution?  Might there be a bug in the error calculation...

> getting an error in some "unprepared" setups
what does it mean?

> Another idea is to add comments to the flavor file. In this case, the sampled arcs can be taken in scope with comments.
could you give me an example?

berkut0, 2022-10-29:
> This is linear approximation imho

Yes, perhaps I didn't put it correctly. I'm talking about interpolation. Sampling and interpolation. Roughly speaking, it would be cool to get rid of the arcs and make some kind of point approximation instead. In some cases this solves a lot of problems where arcs can be an issue

> what does it mean?

Now most of the problems in my setups (which are not originally designed for g-code, but I use them in different ways) are due to arcs and their calculations. Sometimes it's on the machine side, sometimes it's on the processing computer side. One one hand It is much easier to interpret and use a series of points instead of complicated arches from g-code, and on other some machines face the problem of planning movement around such complex motion

> could you give me an example?

Something like this:
``` g-code
; Start Arc Interpolation
G00 X0 Y0 Z0
G00 X1 Y2
G00 X3 Y4
; End Arc Interpolation
```

berkut0, 2022-10-29:
I tuned the settings and I use the "-r" parameter a lot and it helps, but sometimes at the most unlikely moment, the code fails and I have to run everything from the beginning with higher values of "-r"

domoszlai, 2022-10-29:
I think we could generalize some settings into one, like
`--interpolation=linear,biarc,cubic-bezier`
where `biarc` is the default (G03/G04), `cubic-bezier` (G05) we already have and `linear` (G01/G02) could be trivially added. Then `--generate-bezier` would be deprecated 

This would work for you?

domoszlai, 2022-10-29:
I'm more interested in how you get too small values. Could you give me an SVG when `-r` is propery set, and you still get some problems?

berkut0, 2022-10-29:
Sounds great! Thanks a lot!
Would there be an option for sampling level?

berkut0, 2022-10-29:
If I encounter such a situation again, I'll post it here. But I can't always tell exactly what's causing it. The only thing that helps is to manipulate the "-r"

> I'm more interested in how you get too small values. Could you give me an SVG when `-r` is propery set, and you still get some problems?

If I encounter such a situation again, I'll post it here. But I can't always tell exactly what's causing it. The only thing that helps is to manipulate the "-r"

domoszlai, 2022-10-29:
> Would there be an option for sampling level?
I need to think about it. There should be something general, like maximal acceptable error

Iqwertz, 2022-11-03:
> I think we could generalize some settings into one, like `--interpolation=linear,biarc,cubic-bezier` where `biarc` is the default (G03/G04), `cubic-bezier` (G05) we already have and `linear` (G01/G02) could be trivially added. Then `--generate-bezier` would be deprecated
> 
> This would work for you?

This would be a really great option to have! Also an option to set the level of error for the linear approximation would be great.

domoszlai, 2022-11-26:
I just released v0.2.1.0. It completely overhauls how error is calculated. gcode output should be much smaller and more accurate regarding the `resolution` parameter. Please give it a try

berkut0, 2022-12-08:
I think I'll be able to try it soon. In general, only the logic of error handling has changed? But there is no "no arches" option yet?

domoszlai, 2022-12-08:
Look at the branch `linear`. I need to update doc and release it (maybe next week), but you can test it already. It generates way bigger gcode for the same error tolerance and the result is not G1, but at least it proves how good the biarc approximation is 😁

berkut0, 2022-12-08:
does it need to be compiled?

domoszlai, 2022-12-08:
Yep, see instructions in README. Should be straightforward

berkut0, 2022-12-08:
Please give me some clarification
In what ranges should the new parameters be
[image]

domoszlai, 2022-12-08:
linear, biarc, cubic-bezier

domoszlai, 2022-12-09:
Any luck?

berkut0, 2022-12-09:
I'm trying out
But I encounter problems that are not specifically related to Juicy-Gcode
Everything seems to be working. But I will be able to do in-depth tests only later.

berkut0, 2022-12-13:
Actually, linear version works perfectly!
It was possible only because of your tool!
[image]

domoszlai, 2022-12-13:
Nice robot! :) I'll release it later this week

domoszlai, 2022-12-17:
v0.3.0.0 has been released, I close this ticket

closed 2022-12-17 by domoszlai