domoszlai/juicy-gcode #19: Compiled binaries for rpi (armv6hf)

Iqwertz, 2022-11-26:
I want to use juicy-gcode in a project running on an rpi 3b. However the currently provided release binaries are only 64 bit. Would it be possible to also provide armv6hf binaries? I dont have any knowledge on Haskell and Stack and therefore I am not sure if this is possible.

domoszlai, 2022-11-26:
I use `appveyor` to build binaries and they do not support this platform, I also do not own an rpi, so I can't promise it. I think you should try to build the project yourself, it should not be hard. Just install `stack` and try to build the project:

`curl -sSL https://get.haskellstack.org/ | sh`
stack build

I can try to help if you run into problems

Iqwertz, 2022-11-26:
Okay thanks! I alerady tried it, but I ran into memory problems while compiling. But if I find some time I will try to compile it with the rpi 4.