domoszlai/juicy-gcode #20: Control feedrate with object colour

SuperMonkeyRules, 2023-01-11:
Its not really an issue but I think it would be helpful if juicy gcode could be able to get the colour of an object and use that to set that paths feedrate.

Im trying to read through the code and do it myself, but im not very familiar with Haskell.
Any guidance would be great, if you dont want to do it your self.

domoszlai, 2023-01-14:
I've been thinking about using colors as additional input for a while. The simplest thing could be something like a mapping from color to GCODE commands in a config file. The commands could be injected into the output when the color changes... I don't have the time do it right now, but I can help you of course if you have any specififc question

domoszlai, 2023-01-14:
```
gcode {
   begin = "G17;G90;G0 Z1;G0 X0 Y0"
   end = "G0 Z1"
   toolon =  "G00 Z1"
   tooloff = "G01 Z0 F10.00"
}
colors {
   000000 = "M220 S50"
   0000ff = "M220 S80"
}
```

Something like this. Could it work?

domoszlai, 2023-01-26:
So, do you think the proposed solution would work for you? I don't want to invest in a complicated thing before it proved to be necessary. Also, would you be able to test it?

SuperMonkeyRules, 2023-01-26:
It looks like it would work and I would be happy to test it.

domoszlai, 2023-01-28:
Are you able to build the project?

SuperMonkeyRules, 2023-01-28:
Im not too familiar with it but I could try.

domoszlai, 2023-01-29:
All right, let's give a try. It should be pretty easy with `stack`. You should follow the installatin section in the README, but run an extra `git checkout color` after `git clone`.

So, I preperad branch `color`. It should be able to generate color specific gcode. Unfortunately, I had to change the config file format to YAML. Anyway, this is the config I used, I hope it gives you an idea how to use it:

```
begin: 
  - "G17"
  - "G90"
  - "G0 Z1"
  - "G0 X0 Y0"
end:
  - "G0 Z1"
toolon:
  - "G01 Z0 F10.00"
tooloff:
  - "G00 Z1"
colors:
   "000000": 
       - "; black"
   "FF0000": 
       - "; red"
   "FFFF00": 
       - "; yellow"   
```

SuperMonkeyRules, 2023-01-29:
Thank you so much
Ill try it out in the morning.

SuperMonkeyRules, 2023-01-30:
Its able to convert the svg to gcode but doesnt seem to add any thing to do with colours.
This is what I have under config.yaml and Im running the command with ``juicy-gcode.exe drawing.svg --curve-fitting=linear -f config.yaml -o test.gcode``
``
begin: 
  - "G21"
  - "G00 Z1"
  - "G00 X0.0 Y0.0"
end:
  - "G00 Z1"
  - "G00 X0.0 Y0.0"
toolon:
  - "G01 Z0"
tooloff:
  - "G00 Z1"
colors:
   "000000": 
       - "F1.0 S80"
   "FF0000": 
       - "F2.0 S25"
``

domoszlai, 2023-01-30:
I added your config along with a simple SVG + generated output here: https://github.com/domoszlai/juicy-gcode/tree/color/resources/tests/color

I hope helps to eliminate most of the potential problems.

* please double check the config file. it should look like this: https://github.com/domoszlai/juicy-gcode/blob/color/resources/tests/color/config.yml
* do you use stroke color in your SVG? A working example https://github.com/domoszlai/juicy-gcode/blob/color/resources/tests/color/lines.svg

domoszlai, 2023-05-29:
The latest release, v1.0.0.0, enables you to set color specific feedrate. Please have a look and reopen this case if it does not work for you

closed 2023-05-29 by domoszlai