domoszlai/juicy-gcode #23: Flavor as an application argument?

berkut0, 2023-08-28:
Hi! Is it possible to pass flavors as an argument?
Not as a file, but as a straight parameter argument

berkut0, 2023-08-29:
Got it, thanks

berkut0, 2023-08-29:
It seems to me that configuration could be provided in "real time". I'm constructing the configuration in some sort of configurator and pushing that out to the application. And in this pipeline files feels a bit redundant. I think this problem is mostly for the Windows platform.

Maybe it's possible to pass not a file, but pure YAML in quotes instead of a file?

domoszlai, 2023-09-01:
Not sure. Why would this be useful?

berkut0, 2023-09-03:
It's related to https://github.com/domoszlai/juicy-gcode/issues/24
It feels like to be great to have it as a pure CLI tool
Now we should provide files, it's nice, but if I can feed pure SVG as input, maybe even a parts of SVGs, provide flavors as parameters and it can be easily and seamless integrated in very different scenarious

Maybe it's just a fantasy, and to be honest, not as necessary as I can imagine.