domoszlai/juicy-gcode #26: Does colour support have an "after" option?

berkut0, 2023-08-29:
Is that an exhaustive number of settings?
It may also be worth adding a 'default' option in the event of failed detection

``` YAML
colors:
   "000000": 
      before: |
        ; black
      passes: 1
      parameters:
        F: 1.0 
        S: 80
   "FF0000": 
      before: |
        ; red
      passes: 1
      parameters:
        F: 2.0 
        S: 25
```

domoszlai, 2023-09-01:
Probably a "default" option would make sense

berkut0, 2023-09-03:
@domoszlai and what do you think about extending this config a bit to add "after:" for example?

domoszlai, 2023-09-04:
"after" also makes sense, probably. Could you give me an example when it would be useful?

berkut0, 2023-09-05:
Do I have to specify these parameters if they exist in the config?
Can they be optional and not specified if they are not used?

Mostly for consistency, but I want to use "After" in my setup to clean the nozzle of a spray head before changing a tool