domoszlai/juicy-gcode #27: Parametric tokens to use with flavor

berkut0, 2023-09-05:
I use PrusaSlicer a lot. And it has a very unique and powerful feature - it has internal parametric computable tokens. Almost every parameter in the application has a token name. And you can use them in any place of your gcode manipulation: if you want to write your personal parking routine, you can use width of heatbed or thickness of layer. You can even write some kind of script and test parameters changing with a layer height.

I found myself writing something like this:
[image]

Now I'm implementing token replacement and it's working great in my setup. But what if we could use some kind of scripting in our configuration? Width and height of the canvas can be used to write routines to center on image, date, filename, number of commands and comments - first thing that comes to mind

domoszlai, 2023-09-08:
so you can provide custom parameters, not a fixed set coming from the software?

berkut0, 2023-09-08:
Yes, I'm using these tokens to format the final file. Actually I'm not using gcode by itself in the current setup, but using result for parsing to final robot job. And "{0:datefime}" at some points becomes datetime of job parsing.

What I think would be extremely useful with your software - possibility to use some tokens and maybe even calculation on them.
Imagine you could write something like this:
``` gcode
; {date_time} : {file_name}, {flavor_name}
G00 X{canvas_width/2} Y{canvas_height/2}
```
Maybe you shouldn't stop at implementing colours. It would be great to have access to other attributes such as layer names, figure types, widths. The ability to disable, enable or modify parsing based on some data. Maybe you even decide to implement fill hatching?

berkut0, 2023-09-09:
[image]