domoszlai/juicy-gcode #28: cannot build with stacks

monperrus, 2023-10-11:
hi  @domoszlai thanks a lot for your work on juicy-gcode.

however, build does not work on my machine.

any suggestion to make progress?

Thanks!

```
$ stack --version
2.3.3 x86_64

$ stack build
Stack has not been tested with GHC versions above 8.10, and using 9.4.7, this may fail
Preparing to install GHC (tinfo6) to an isolated location.
This will not interfere with any system-level installation.
Already downloaded.                       
Unpacking GHC into /home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7.temp/                                                                                     Did not find executable at specified path: /home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7.temp/ghc-9.4.7/configure
                   
Error: Error encountered while configuring GHC with
         /home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7.temp/ghc-9.4.7/configure --prefix=/home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7/
         run in /home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7.temp/ghc-9.4.7/
       
       The following directories may now contain files, but won't be used by stack:
         - /home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7.temp/
         - /home/martin/.stack/programs/x86_64-linux/ghc-tinfo6-9.4.7/
       
       For more information consider rerunning with --verbose flag
       
Configuring GHC ...
```

```
$ stack --version
2.3.3 x86_64
Compiled with:
- Cabal-3.0.1.0
- Glob-0.10.1
- StateVar-1.2
- aeson-1.4.7.1
- annotated-wl-pprint-0.7.0
- ansi-terminal-0.10.3
- ansi-wl-pprint-0.6.9
- array-0.5.4.0
- asn1-encoding-0.9.6
- asn1-parse-0.9.5
- asn1-types-0.3.4
- async-2.2.2
- attoparsec-0.13.2.4
- attoparsec-iso8601-1.0.1.0
- auto-update-0.1.6
- base-4.13.0.0
- base-compat-0.11.1
- base-compat-batteries-0.11.1
- base-orphans-0.8.2
- base16-bytestring-0.1.1.7
- base64-bytestring-1.0.0.3
- basement-0.0.11
- bifunctors-5.5.7
- binary-0.8.7.0
- bitarray-0.0.1.1
- blaze-builder-0.4.1.0
- blaze-html-0.9.1.2
- blaze-markup-0.8.2.7
- bytestring-0.10.10.1
- casa-client-0.0.1
- casa-types-0.0.1
- case-insensitive-1.2.1.0
- cereal-0.5.8.1
- clock-0.8
- colour-2.3.5
- comonad-5.0.6
- conduit-1.3.2
- conduit-extra-1.3.5
- connection-0.3.1
- constraints-0.12
- containers-0.6.2.1
- contravariant-1.5.2
- cookie-0.4.5
- cryptohash-sha256-0.11.101.0
- cryptonite-0.26
- cryptonite-conduit-0.2.2
- data-default-class-0.1.2.0
- deepseq-1.4.4.0
- digest-0.0.1.2
- directory-1.3.6.0
- distributive-0.6.2
- dlist-0.8.0.8
- easy-file-0.2.2
- echo-0.1.3
- ed25519-0.0.5.0
- either-5.0.1.1
- enclosed-exceptions-1.0.3
- exceptions-0.10.4
- extra-1.7.4
- fast-logger-3.0.1
- file-embed-0.0.12.0
- filelock-0.1.1.5
- filepath-1.4.2.1
- fsnotify-0.3.0.1
- generic-deriving-1.13.1
- ghc-boot-th-8.8.4
- ghc-prim-0.5.3
- hackage-security-0.6.0.1
- hashable-1.3.0.0
- hi-file-parser-0.1.0.0
- hinotify-0.4
- hourglass-0.2.12
- hpack-0.34.2
- hpc-0.6.0.3
- http-api-data-0.4.1.1
- http-client-0.6.4.1
- http-client-tls-0.3.5.3
- http-conduit-2.3.7.3
- http-download-0.2.0.0
- http-types-0.12.3
- infer-license-0.2.0
- integer-gmp-1.0.2.0
- integer-logarithms-1.0.3
- libyaml-0.1.2
- lifted-async-0.10.1.2
- lifted-base-0.2.3.12
- megaparsec-8.0.0
- memory-0.15.0
- microlens-0.4.11.2
- microlens-mtl-0.2.0.1
- microlens-th-0.4.3.5
- mime-types-0.1.0.9
- monad-control-1.0.2.3
- monad-logger-0.3.34
- monad-loops-0.4.3
- mono-traversable-1.0.15.1
- mtl-2.2.2
- mustache-2.3.1
- neat-interpolation-0.3.2.6
- network-3.1.1.1
- network-uri-2.6.3.0
- old-locale-1.0.0.7
- old-time-1.1.0.3
- open-browser-0.2.1.0
- optparse-applicative-0.15.1.0
- pantry-0.4.0.2
- parsec-3.1.14.0
- parser-combinators-1.2.1
- path-0.7.0
- path-io-1.6.0
- path-pieces-0.2.1
- pem-0.2.4
- persistent-2.10.5.2
- persistent-sqlite-2.10.6.2
- persistent-template-2.8.2.3
- pretty-1.1.3.6
- primitive-0.7.0.1
- process-1.6.9.0
- profunctors-5.5.2
- project-template-0.2.1.0
- random-1.1
- regex-applicative-0.3.3.1
- regex-applicative-text-0.1.0.1
- resource-pool-0.2.3.2
- resourcet-1.2.4.2
- retry-0.8.1.2
- rio-0.1.18.0
- rio-orphans-0.1.1.0
- rio-prettyprint-0.1.1.0
- rts-1.0
- safe-0.3.19
- safe-exceptions-0.1.7.0
- scientific-0.3.6.2
- semigroupoids-5.3.4
- semigroups-0.19.1
- shelly-1.9.0
- silently-1.2.5.1
- socks-0.6.1
- split-0.2.3.4
- stm-2.5.0.0
- stm-chans-3.0.0.4
- streaming-commons-0.2.2.1
- syb-0.7.1
- tagged-0.8.6
- tar-0.5.1.1
- tar-conduit-0.3.2
- template-haskell-2.15.0.0
- temporary-1.3
- text-1.2.4.0
- text-metrics-0.3.0
- th-abstraction-0.3.2.0
- th-expand-syns-0.4.6.0
- th-lift-0.8.1
- th-lift-instances-0.1.17
- th-reify-many-0.1.9
- time-1.9.3
- time-compat-1.9.3
- tls-1.5.4
- transformers-0.5.6.2
- transformers-base-0.4.5.2
- transformers-compat-0.6.5
- type-equality-1
- typed-process-0.2.6.0
- unicode-transforms-0.3.6
- unix-2.7.2.2
- unix-compat-0.5.2
- unix-time-0.4.7
- unliftio-0.2.13
- unliftio-core-0.1.2.0
- unordered-containers-0.2.10.0
- uuid-types-1.0.3
- vector-0.12.1.2
- vector-algorithms-0.8.0.3
- x509-1.7.5
- x509-store-1.6.7
- x509-system-1.6.6
- x509-validation-1.6.11
- yaml-0.11.4.0
- zip-archive-0.4.1
- zlib-0.6.2.2

```

domoszlai, 2023-10-11:
isn't it just an old `stack`? I have 2.9.1

domoszlai, 2023-10-11:
try https://www.haskell.org/ghcup/

monperrus, 2023-10-12:
this is the default ubuntu stack. I'll try with a more recent one.
thanks!

closed 2023-10-12 by monperrus

monperrus, 2023-10-13:
complete success today with juicy-gcode today, thanks a lot @domoszlai for the great tool