domoszlai/juicy-gcode #29: -d option is not considered when svg height and width are length in centimeters

monperrus, 2023-11-11:
Given a SVG file with height and width as real length in centimeters
```svg
    <svg height="26cm" version="1.2" viewBox="0 0 500.0 500.0" width="26cm">
```

### Actual behavior

 the -d DPI option is silently ignored

### Expected behavior

Either
* a warning is given stating "`-d` option ignored because SVG height and width are in centimeters"
* the program stops with error message "`-d` conflicts with SVG height and width in centimeters"

domoszlai, 2023-11-11:
You do not need the DPI in this case as 26cm is ~10in so the SVG has ~50DPI. This reflected in `juicy-gcode --help`: 

<img width="580" alt="image" src="https://github.com/domoszlai/juicy-gcode/assets/8457752/733a4524-774b-4be0-
a160-6dbad4ebace8">

I feel like a warning would be appropriate, but also luxurious :D

domoszlai, 2023-11-11:
Actually, there might be a bug in the current implementation:

```SVG
<svg width="10cm" height="10cm" viewBox="0 0 200 200" xmlns="http://www.w3.org/2000/svg">
  <rect x="1cm" y="1cm" width="10" height="20"/>
</svg>
``` 

Not sure in this case what should happen...

domoszlai, 2023-11-11:
Looking at this example: 

```SVG
<svg width="10cm" height="10cm" viewBox="0 0 300 300" xmlns="http://www.w3.org/2000/svg">
  <rect x="1in" y="1in" width="10" height="20"/>
  <rect x="96" y="96" width="10" height="20"/>
</svg>
````

The two rectangles rendered to the same spot independently of what the `viewBox` is. Thus, some default DPI is used to do the inner conversion, event though, the document size is in cm. This is exactly how `juicy-gcode` works, so the DPI setting still required

monperrus, 2023-11-12:
I confirm that with this example, `-d` has an effect on the first rectangle.

Now coming back to the use case.

I was changing `-d` without any effect on the output. After a tedious debug session, I realized that the input SVG had something new: the real length.

What would be the luxurious message to save me precious time?

Maybe "Drawing area {X}cm x {Y}cm onto gcode max_x=100,max_y=100"

 Thanks!

domoszlai, 2023-11-13:
But as my previous example suggests, even if the width and height of the SVG is in cm, you might need the DPI to render it. 
So I can't give a warning like "-d option ignored because SVG height and width are in centimeters", because it is not true (it might not being used but it is not ignored). 

I also can't just dump debug messages to the standard output freely as I might break other people's application.

I could probably add like a debug or info flag...

monperrus, 2023-11-14:
ack, thanks!